openwrt/firmware-selector-openwrt-org #6: Checksums for OpenWRT 22.03.5 for Espressobin v7 sdcars seem to be incorrect

paulmsander, 2024-03-07:
I downloaded these images using Firefox on a Mac, and the SHA-256 checksums of the downloaded files don't match those shown on the download page on the downloads.openwrt.org site.

I also downloaded the factory image for the TP-Link Archer C7, and its checksum matched.

I'm using "shasum -a 256 -b" on the Mac to compute the checksums, and I'm comparing them visually.

paulmsander, 2024-03-07:
It was not obvious that the download image is in gzip format, and it turns out that Safari decompresses downloaded files automatically by default.  The uncompressed files had different checksums, and gzipping them again also produced different checksums.  But downloading the files using a different browser (Firefox) downloaded the .gz files without further processing, and the checksums matched.

mwarning, 2024-03-07:
@paulmsander glad you figured out the issue :-)

closed 2024-03-07 by mwarning