openwrt/firmware-selector-openwrt-org #7: Please allow to build 23.05-SNAPSHOT images

lukasz1992, 2024-03-22:
Hi,

that works but is not exposed to UI.

mwarning, 2024-03-22:
@aparcar do we want to expose those images?

aparcar, 2024-03-22:
I'd do it but @jow- was a bit concerned. Maybe we should add big warnings or something?

mwarning, 2024-03-22:
I do not mind either way.

efahl, 2024-04-24:
I wondered about this a couple days before I stumbled into this issue.  I had chased it to this line:

https://gitlab.com/openwrt/web/downloads-openwrt-org/-/blob/main/dir-index.cgi?ref_type=heads#L395

(Side quest: Is 'downloads' source still on gitlab, or did it migrate over here on github like firmware selector did?)

My opinion is that they should be exposed, as at least every week there's a post on the forums on the order of "if you want the latest patches and can't wait for the next release, just install xx.xx-snapshot" assuming that the reader will be able to figure out where they live...

mwarning, 2024-04-24:
@efahl I think it is up to @aparcar and @jow-.

efahl, 2024-11-15:
Both `auc` and `owut` already expose release snapshots (owut does it explicitly, `owut versions` shows all of the available release snapshots).  The FS code already has that switch we could throw to expose them: https://github.com/openwrt/firmware-selector-openwrt-org/blob/master/www/index.js#L828, so it's simply setting that boolean in the config.js.

The missing piece for 24.10 is the tag in the main repo, needed to trigger creation of an entry in https://downloads.openwrt.org/.versions.json

efahl, 2025-01-19:
> but not (yet) 24.10-SNAPSHOT (as of 2024-11-28).

Yeah, the upcoming version was being ignored.  See https://github.com/openwrt/firmware-selector-openwrt-org/pull/68, should fix it...

mwarning, 2025-02-17:
Closing this as this seem to have been rejected for https://firmware-selector.openwrt.org/
But selecting *-SNAPSHOT images is possible via https://openwrt.github.io/firmware-selector-openwrt-org/

closed 2025-02-17 by mwarning