openwrt/firmware-selector-openwrt-org #20: allow defaulting to snapshots builds if no stable builds are found for a device

Lanchon, 2024-07-20:
when using the fw selector, you can remove the `version=<x.x.x>` url parameter to create a link to the latest available build for a device.

for example, this link will pre-select the latest build for WHW03 V2:

- https://firmware-selector.openwrt.org/?target=ipq40xx%2Fgeneric&id=linksys_whw03v2

however, if there are no stable builds for a device yet (still on snapshot), the same type of link will fail:

- https://firmware-selector.openwrt.org/?target=ipq40xx%2Fgeneric&id=linksys_whw03

these links are useful and are sometimes used in devices wikis (and i think they should be auto-generated in the ToH now). however, currently they can't be used for devices still on snapshot (whose wikis tend to be more actively maintained, paradoxically).

please consider fixing this by defaulting to snapshots if stable builds are not found.

mwarning, 2024-09-05:
A missing version in the link is undefined behavior. I like to keep selector simple and not added any logic behind it.

Lanchon, 2024-09-07:
that's too bad... it's reasonable to have a link to the latest release software for each device.

such functionality i baked in many platforms, eg: https://github.com/openwrt/openwrt/releases/latest

in any case, because links missing the version currently resolve to latest version instead of faulting, people reasonably thought that that was feature and started using it. so now there are a bunch of links from device pages to the firmware selector that work but are technically undefined.

for example, [this page](https://openwrt.org/inbox/toh/arris/tr4400_v2) includes this line:

> Official firmware: https://firmware-selector.openwrt.org/?target=ipq806x%2Fgeneric&id=arris_tr4400-v2

yes, that's a page i built myself for a port that i did, but i took inspiration/copied from other wiki entries.

besides that case, i know i added those types of links a couple of times in the wiki, for devices i ported and also others i helped document. i always thought that having to use a special form for devices not yet with a stable (to point to snapshot) release was a nuisance.

mwarning, 2024-09-08:
I had the same idea actually and added it in 4.0.0 (https://github.com/openwrt/firmware-selector-openwrt-org/blob/master/CHANGELOG.md?plain=1#L40)

But the translation layer was removed in favor of simplicity and reducing local state. That makes it very hard for the OFS to implement this. The OFS would need to load all release information at once, that would multiply the OFS load traffic by 30x).