openwrt/firmware-selector-openwrt-org #21: do not update the browser's URL on a simple page load

Lanchon, 2024-07-20:
when you open a link such as this one:

- https://firmware-selector.openwrt.org/?target=ipq40xx%2Fgeneric&id=linksys_whw03v2

the page immediately rewrites browser's URL adding `version=23.05.4&` to it. this is a very minor annoyance, but IMHO the page should only modify the browser's URL if the user actually selects something. please consider fixing this.

mwarning, 2024-09-05:
The missing version in the link is actually an oversight that was fixed. But I understand that it might be behavior against the norm/expectation. I will fix it if does not complicate the code too much.

mwarning, 2025-02-08:
Will be fixed here: https://github.com/openwrt/firmware-selector-openwrt-org/pull/81

mwarning, 2025-02-23:
Merged, will be in the next release.

closed 2025-02-23 by mwarning