openwrt/firmware-selector-openwrt-org #22: implement a way to easily add some very comonly used packages

Lanchon, 2024-07-20:
there a handful of packages the are added constantly to images because they are key to enable bootstrapping a device to the point where more packages can easily be installed. IMHO, these are:

- `luci`: allows regular users to easily install packages and to connect to the internet if the upstream connection is anything but a straight ethernet cable.
- `luci-ssl`: some users need SSL to configure their router securely. (also added for completeness.)
- `luci-app-attendedsysupgrade`: if you require or are forced to use a snapshot, after some time ASU becomes the only option to add certain packages (eg, kernel modules). (also, LuCI ASU is the only way to easily add packages to a device that is not connected to the internet --just as long as the browser you are using is.)
- `owut`: as above, but some users prefer the command line. (also added for completeness.)

it'd be great to have an easy (point and click) way to add at least these packages once "Customize installed packages and/or first boot script" is unfolded.

one option would be to have checkboxes to add these packages to the build once the build is requested, but this might be too much work.

the easiest way would be to add a gear icon to the package list text box (just like in the box below it), and when you click it it just adds text to the text box.

this strategy would be enough: for each package to add, regex the package name (full word) in the text of the text box, and if not found add "/n${PACKAGE_NAME}" to the text. the CR is valid whitespace in this text, and the line breaks would clearly show what was added (and let you easily remove what is unwanted).

final note: `owut` recently replaced `auc`. IMHO, you can simply add `owut` be done with it. `auc` will soon enough go unused, and its name is short enough that typing it is not a problem. however if you really want to nit pick, you can use `auc` for versions 23.x and lower, and `owut` for the rest, including snapshot.

mwarning, 2024-09-05:
We already have an option to automatically add packages to the package list and it is used to add the luci package: https://github.com/openwrt/firmware-selector-openwrt-org/blob/f38d80f4788dce869d5069b5666b1f10e4653b73/www/config.js#L27

Isn't that already fullfilling your request?

Lanchon, 2024-09-07:
@mwarning 

> Isn't that already fullfilling your request?

no... :(

what i mean is extending the UI in https://firmware-selector.openwrt.org/ to allow point-and-click adding of the very highly used packages i mentioned above. as said, those packages are key to being able to install other packages later on.

mwarning, 2024-09-07:
I have to say that this fine level of detail is probably out of the scope of the firmware selector. But I let @aparcar decide, since this is an ASU related feature. :-)

Lanchon, 2024-11-30:
> Can we also add `luci-app-sqm` to the list?

IMHO, definitely not. the packages i proposed are exclusively packages that enable users to install other packages.

a much better UI could be provided with live search of packages, such as the one in LuCI/System/Software, but this is outside the scope of this issue which aims to be simple.