openwrt/firmware-selector-openwrt-org #33: Error when create new custom firmware!

hainguyen-cmd, 2024-09-05:
I have tried to create my custome firmware with some pre-installed packages! However, https://firmware-selector.openwrt.org/ return no action or information after I cliked [REQUEST BUILD]! After trying to know what happening, I have found out that error:
index.js:118 
        
        
       POST https://sysupgrade.openwrt.org/api/v1/build net::ERR_ABORTED 502 (Bad Gateway)

So please solving this! Thanks so much!

hainguyen-cmd, 2024-09-05:
It's my fault DNS setting! So sorrry about this mistake!

hainguyen-cmd, 2024-09-06:
Actually, it is not faulty of network! Maybe https://sysupgrade.openwrt.org has problem!

tskarthikkumar, 2024-09-06:
+1 

I'm also having the same issue when trying to request build. nothing happens after request build button is clicked.

greem, 2024-09-06:
@aparcar firmware selector / ASU server/s are down - maybe not down, but not working.

navigos, 2024-09-06:
Yes. Same here. And it's been like this for some time now.

lindevel, 2024-09-07:
```
$ curl -D - https://sysupgrade.openwrt.org/
HTTP/2 500 
access-control-allow-credentials: true
access-control-allow-headers: *
access-control-allow-methods: POST, GET, OPTIONS
access-control-allow-origin: *
access-control-expose-headers: Authorization
alt-svc: h3=":443"; ma=2592000
content-type: text/plain; charset=utf-8
date: Sat, 07 Sep 2024 06:23:16 GMT
server: Caddy
server: uvicorn
vary: Origin
content-length: 21

Internal Server Error
```

mwarning, 2024-09-07:
Yeah, no idea why the backend is down. But this concerns the backend (ASU: https://github.com/openwrt/asu) and not the frontend, the OpenWrt Firmware Selector.

As such I will close this issue. Please report on the ASU project page or in the OpenWrt forum. Thank you for your concerns.

closed 2024-09-07 by mwarning