openwrt/firmware-selector-openwrt-org #45: Cannot build snapshot image due to typo and incorrect version errors

this-username-has-been-taken, 2024-09-27:
Hello!

I have faced with two issues while trying to build a snapshot version of the OpenWrt.

The first issue is a typo in the configurations list: the firmware selector shows "SNAPSHOT**S**" instead of "SNAPSHOT" resulting in the "Unsupported branch: SNAPSHOTS" error:
[image]

This issue could be easily overcame by opening a browser's developer console and changing the first element ("SNAPSHOTS") in the versions selector to "SNAPSHOT":
[image]

Another issue however is more severe: if I fix the typo and ask for a build then in a few moments I will get "Error: Received incorrect version XXX (requested YYY)" error which can't be easily fixed.

[image]

I kindly ask to fix these two issues. They are blocking all the users who is forced or wants to use snapshot versions from obtaining a custom build.

Steps to reproduce:

1. Open the firmware selector: https://firmware-selector.openwrt.org/?version=SNAPSHOTS&target=mediatek%2Ffilogic&id=bananapi_bpi-r4
2. Click on the "Customize installed packages and/or first boot script" link
3. Click on the "Request build" button
4. Observe the first issue
5. Open the browser's developer console and fix the typo in the versions list as mentioned above
6. Click on the "Request build" button again and wait a bit
7. Observe the second issue

mwarning, 2024-09-27:
Please try again and let us know if it is fixed.

this-username-has-been-taken, 2024-09-27:
I have just checked: the typo is fixed, but unfortunately, the other error: "Error: Received incorrect version XXX (requested YYY)" still present.
I have checked for "Banana BPI-R4" and a few random models - no luck. I have also cleaned the browser cache and tried the private mode (just in case).

aparcar, 2024-09-27:
Yea I'm on it. It's a docker/cache/whatever issue which been annoying for month. Gotta tear down that CI madness and implement it differently. fml.

this-username-has-been-taken, 2024-09-27:
I know your pain. Once a CI/CD is working it is OK. But if suddenly it crushes - you're just sitting in the middle with a bunch of what's left trying to figure out the failure point and what has happened. Too many "moving parts"... Anyway wish you luck in sorting the issue out.

this-username-has-been-taken, 2024-09-27:
I have just tried building a firmware for Banana BPI-R4 and it succeeded. Building for Xiaomi Mi Router AX3000T still fails.
Don't know if anything has been changed, but it might help in resolving the issue.

TurtlePL, 2024-11-06:
it doesn't work again...
"Received incorrect version" or/and "make: *** No rule to make target 'info"

TurtlePL, 2024-11-06:
Error: Received incorrect version XXX (requested YYY)" still present...

mwarning, 2025-02-08:
This bug seems to be fixed. Please reopen if the bug is still present.

closed 2025-02-08 by mwarning