openwrt/firmware-selector-openwrt-org #53: Download Link Opens Binary File in Browser Instead of Prompting Download (Custom Images)

0nelight, 2024-10-26:
Download-Links to freshly build Images via the Browser-Interface of the public [Firmware-Selector-Server](https://firmware-selector.openwrt.org/) on the [public ASU](https://sysupgrade.openwrt.org/) seem to not being correctly advertised as `Content-Disposition: attachment` from the Webserver running on ASU.

This leads to the Issue where, when clicking on the Download-Button, instead of Firefox opening the Download-File-Dialog, it interprets the File as a Webpage and tries to render it. Workaround is clicking "Save Target as".

I am not sure if I should put this here or at ASU - therefore I created a [Ticket there too](https://github.com/openwrt/asu/issues/1059).

mwarning, 2024-10-26:
I cannot reproduce the problem. With Firefox 131.0.3 the generated images are downloaded as octet-stream.
[image]

0nelight, 2024-10-26:
Hmm.. strange - for what device you produced your image? You applied a custom config? 

On my place here I tried with raspi3 device and set rootpw and ip and I get as mime in firefox and in chrome "text". Chrome then does some stuff to get this right anyway, Firefox takes "text" as "text". When I download with "save file as" the image works flawlessly.

0nelight, 2024-10-26:
For "8devices Lima" you are right.. there the Type is correct octet-stream:
[image]

But for "Raspberry Pi 3B" it is not:
[image]

I suspect the "gz" compression is the reason. Maybe related to [this Issue](https://github.com/openwrt/asu/issues/644)?

AliveDevil, 2024-10-26:
Context chain:
- Set up staticfiles: https://github.com/openwrt/asu/blob/main/asu/main.py#L39
- staticfiles from fastio is forwarded to starlette: https://github.com/fastapi/fastapi/blob/master/fastapi/staticfiles.py
- starlette returns FileResponse: https://github.com/encode/starlette/blob/master/starlette/staticfiles.py#L178
- Which, if MediaType is None (which it possibly always is), guesses the content type: https://github.com/encode/starlette/blob/master/starlette/responses.py#L301-L302
- Which is a bundled component: https://docs.python.org/3/library/mimetypes.html

All that is to say: ASU should probably register `mimetypes.add("application/octet-stream", ".img.gz")`.

mwarning, 2024-10-28:
Thank you for figuring this out. Let me close this ticket and hope this will get fixed on the ASU side.

closed 2024-10-28 by mwarning