openwrt/firmware-selector-openwrt-org #55: Unable to Include uboot-envtools in Xiaomi Mi Router 3G Builds

ztj1993, 2024-11-08:
The xiaomi_mi-router-3g model seems to be forcibly excluding the uboot-envtools package. Despite multiple attempts, I have not succeeded in adding uboot-envtools to the build.

I tested this by manually running make image, and the package could be added to the pre-installed packages. I also verified this functionality on the actual device.

I am unsure which code needs modification to resolve this. I found relevant code here: https://github.com/openwrt/openwrt/blob/ca53f2d430ce1f9ff5a560291de1e93380963417/target/linux/ramips/image/mt7621.mk#L3103, but I am not certain if submitting a PR with modifications to this file is correct.

I hope this issue can be addressed.

mwarning, 2024-11-09:
The reason might be here:
https://github.com/openwrt/openwrt/commit/79bd0172ca5671f7cdf382dc80ec1630da057947

`uboot-envtools` added to DEFAULT_PACKAGES here and force removed in DEVICE_PACKAGES (I assume that is what the `-` prefix does). I wonder why it was done that way.

mwarning, 2024-11-09:
Maybe some dev clan clear that up?
@hauke do you know why `uboot-envtools` is forcefully removed from `xiaomi_mi-router-3g`?

ztj1993, 2024-11-09:
I noticed the commit https://github.com/openwrt/openwrt/commit/79bd0172ca5671f7cdf382dc80ec1630da057947, but I didn't submit a PR due to unfamiliarity with the process.

In versions 23.05.0 through 23.05.3, adding `uboot-envtools` to the default build for the xiaomi_mi-router-3g caused errors, although it could be installed via `opkg install uboot-envtools`. The exact reason is unclear.

I noticed the exclusion of `-uboot-envtools` in the https://downloads.openwrt.org/releases/23.05.5/targets/ramips/mt7621/profiles.json file, and `/api/v1/build` also forces the exclusion of `uboot-envtools`.

Building locally with `openwrt-imagebuilder-23.05.5-ramips-mt7621.Linux-x86_64.tar.xz` did not result in any issues.

I hope someone familiar with the process can solve this issue. I'm not sure if modifying line 3103 in this file https://github.com/openwrt/openwrt/blob/ca53f2d430ce1f9ff5a560291de1e93380963417/target/linux/ramips/image/mt7621.mk#L3103 to include `-uboot-envtools` will resolve it. Perhaps a more standardized approach is needed to handle this.

ztj1993, 2024-11-09:
I removed `-uboot-envtools` from `Installed Packages` and added `uboot-envtools`. ASU shouldn't be removing this package.

Changing `diff_packages` to `false` in the submission to the `/api/v1/build` interface solves this issue, but there is no option to set `diff_packages = false` in the firmware-selector-openwrt-org UI.

mwarning, 2024-11-10:
Maybe it is a bug in the ASU backend. The selector does no logic there.

mwarning, 2024-11-17:
This is probably a ASU bug, as such I suggest to report this here https://github.com/openwrt/asu/

closed 2024-11-17 by mwarning