openwrt/firmware-selector-openwrt-org #58: config: enable release snapshots in selector

efahl, 2024-11-28:
@mwarning @aparcar 

What do you think about toggling on the release-SNAPSHOT builds in FS?  They are already available in `auc` and `owut` by default, seems like it's just a roadblock for people who want to use FS to do the same job.

https://github.com/openwrt/firmware-selector-openwrt-org/blob/master/www/config.js#L17

aparcar, 2024-11-28:
It's hidden so no regular user breaks the system thinking it's fine to install something unreleased.

Feel free to use the GitHub page here with snapshots enabled, however it does not show 24.10 since it not in the `.versions.json` file.

https://openwrt.github.io/firmware-selector-openwrt-org/

efahl, 2024-11-28:
Oh, yeah, I forgot about that one.

efahl, 2024-11-28:
Ok, so how about adding them as `Danger-24.10-SNAPSHOT` and so on as their labels, then sort the list so they are at the bottom.  Also add a comment about "Release SNAPSHOT builds are potentially dangerous, use at your own risk".  That seems like it would be good enough to instill some hesitation into people so they do their homework before blindly installing them.

mwarning, 2024-11-28:
How about we add  https://openwrt.github.io/firmware-selector-openwrt-org/ to the github project description? This has snapshots enabled. But normal users should not be exposed to snapshots.

efahl, 2024-11-29:
Sure, that would be good enough for me.  Add a bit of verbiage "advanced users blah blah snapshots can brick your device..."

efahl, 2024-11-30:
> Why is 24.10-SNAPSHOT not listed then

When the FS `show_snapshots` config setting is enabled, it scans the `.versions.json` input for major versions, and synthesizes a new `xx.yy-SNAPSHOT` from what's there.  Since there's no 24.10 version in `.versions.json` yet, it can't know to create a snapshot for it.  

https://github.com/openwrt/firmware-selector-openwrt-org/blob/master/www/index.js#L789

Once `24.10.0-rc1` is published, the `24.10-SNAPSHOT` will magically appear.

mwarning, 2024-12-01:
OK, then let's add the link to the firmware selector mit the snapshots somewhere (https://openwrt.github.io/firmware-selector-openwrt-org/) and close this ticket. It is important for the common user not to the tempted or confused to download release snapshots.

efahl, 2024-12-02:
If that's the general feeling, then I'd say take main SNAPSHOT out of the public firmware selector, too, as it is generally far more dangerous than the release snapshots.

mwarning, 2024-12-02:
@efahl true, SNAPSHOT is far more dangerous. But at least it is not that easy to be mistaken as something stable.
But I am not inclided to discuss things I cannot change. :-)

efahl, 2025-01-19:
Resolved in https://github.com/openwrt/firmware-selector-openwrt-org/issues/58#issuecomment-2510268676

closed 2025-01-19 by efahl