openwrt/firmware-selector-openwrt-org #61: https://sysupgrade.openwrt.org/api/v1/build not working on localhost and website

GermanAizek, 2024-12-06:
out of interest, I changed my IP, thinking that there are restrictions on ISP provider or server, and the same thing. How do I build an image with a convenient web panel?

```
index.js:868 versions: 23.05.5,19.07.10
index.js:131 
        
        
       POST https://sysupgrade.openwrt.org/api/v1/build net::ERR_ADDRESS_UNREACHABLE
buildAsuRequest @ index.js:131
(anonymous) @ VM26:1
index.js:91 Uncaught (in promise) TypeError: message.startsWith is not a function
    at showStatus (index.js:91:24)
    at index.js:192:7
showStatus @ index.js:91
(anonymous) @ index.js:192
Promise.catch
buildAsuRequest @ index.js:191
(anonymous) @ VM26:1
```

AreYouLoco, 2024-12-07:
https://stackoverflow.com/questions/38813040/startswith-is-not-a-function-getting-called-for-no-reason

AreYouLoco, 2024-12-07:
I've tried on Firefox on Android 133.0.0 as well as Chrome 131.0.6778.104
Chrome throws init error and Firefox same error as in the original Issue description.

GermanAizek, 2024-12-07:
> @GermanAizek hm, what is your brower and version?

Im tested on:
- LibreWolf 133.0-3
- Chromium Version 131.0.6778.108 (Official Build) built on Debian GNU/Linux trixie/sid (64-bit)

mwarning, 2024-12-08:
This might be an ASU issue.
@aparcar do you know?

AreYouLoco, 2024-12-10:
Fixed for me. Also new 24.10-rc2 works now

closed 2025-01-21 by mwarning