openwrt/firmware-selector-openwrt-org #86: expand image size

hadifarnoud, 2025-02-17:
I was hoping I can build an expanded image with this firmware generator. is it possible?

I tried everything else for expanding the SD. It seems to corrupt the filesystem on reboot. 
this would be an awesome option if it does not exist.

hadifarnoud, 2025-02-17:
btw it will also throw an error if I put too many packages to install because it runs out of space. it would be useful to expand the image size for this reason as well

mwarning, 2025-02-17:
hi, that idea was suggested before. But the problem is that the backend (Attended SysUpgrade server) needs to support this.
The firmware-selector is merely front-end for building images.

Please ask  here: https://github.com/openwrt/asu

closed 2025-02-17 by mwarning

hadifarnoud, 2025-02-18:
thanks for the info. I will ask there. it's a great feature to have though. wouldn't you agree?

mwarning, 2025-02-18:
Sure, it would be nice to have. But I do not know if it is useful enough to put in the firmware-selector.

hadifarnoud, 2025-02-18:
in the extra options section, wouldn't hurt to have. it's a pain to expand ext4. especially if you're on windows

aparcar, 2025-02-18:
In the API is a property documented you can just send it along. I think you can simply add an integer field to the OFS request form