openwrt/firmware-selector-openwrt-org #99: Unable to customize packages for 21.02.x - init message

energy-x, 2025-04-16:
I am trying to customize an image to test an upgrade of an older version (21.02.2) and no matter what packages I add or remove from the list, it stops the process with a red "init" after "Received Build Request" in the status. Are these older versions not customizable?

[image]

mwarning, 2025-04-16:
I can confirm this issue. This is a backend issue.
@aparcar can you take a look?

efahl, 2025-04-17:
Sounds like same root cause as this one: https://github.com/openwrt/docker/issues/171

We need to fix up FS to give a better message for this error...

mwarning, 2025-04-17:
I have made a MR to show the error instead of "init":
https://github.com/openwrt/firmware-selector-openwrt-org/pull/100

efahl, 2025-06-20:
Just tested this with https://github.com/openwrt/asu/pull/1432 and it now instead of `init`, it shows

`Image not found: ghcr.io/openwrt/imagebuilder:x86-64-v21.02.2`

so, yeah, we can't build old versions due to missing docker containers.

Above PR addresses the situation where spurious "Internal server errors" occur during container setup, no details are reported by the ASU server.

gyzerok, 2025-09-13:
I am getting the same "init" error while trying to build 24.10.2. Should I open a separate issue?

[image]

efahl, 2025-09-13:
> I am getting the same "init" error

No, the ASU server is currently down.  This is a known issue.

efahl, 2025-09-13:
@mwarning What do you think about changing to `default:` right here?

https://github.com/openwrt/firmware-selector-openwrt-org/blob/c9b18bfbc6208642a1effcb96919eb22684cab89/www/index.js#L168

I think I've seen some errors not in that 400, 422, 500 set...  It would be nice to just grab everything that's unexpected instead of trying to list them all.

mwarning, 2025-09-13:
@efahl I am happy with replacing those cases with default. :)

gyzerok, 2025-09-14:
> > I am getting the same "init" error
> 
> No, the ASU server is currently down. This is a known issue.

Any place I can subscribe to for status? Unfortunately got myself a bit bricked, so need a fresh reinstall

efahl, 2025-09-14:
> Any place I can subscribe to for status?

I've been posting my findings here, so that's sort of the "source of record": https://github.com/openwrt/asu/issues/1488

closed 2025-09-15 by aparcar