openwrt/firmware-selector-openwrt-org #102: Internal Server Error (code 500) on POST /build after Request Build

ronjouch, 2025-04-18:
Looks similar to #88 from back in February. Ping @mwarning @aparcar.

[image]

ronjouch, 2025-04-18:
Might also be related to #99, but I don’t get the red `init` message, so that’s why I created a new bug.

mwarning, 2025-04-18:
Hi, I think this got fixed along with https://github.com/openwrt/firmware-selector-openwrt-org/issues/88
But the changes are not deployed yet.

Can this eror still be triggered (and how)? Then I can check it with latest changes.

ronjouch, 2025-04-19:
@mwarning yeah, sorry for the lazy issue with vague screenshot and no STR.

**Steps to repro:**

1. https://firmware-selector.openwrt.org/?version=24.10.1&target=mediatek%2Ffilogic&id=glinet_gl-mt6000
2. Click `Customize installed packages and/or first boot script`
3. In the `Installed Packages` text box, add ` adblock luci-app-adblock luci-app-sqm diffutils`
4. Click `Request Build`

**Expected**: it builds

**Actual, 2h ago**: `Internal Server Error` (see top screenshot)

**Actual, now (2025-05-18 23:19 EST)**: I no longer get an `Internal Server Error`... but now it looks like the server-side build queue is stalled: Looking at DevTools/Network, the periodical requests `GET https://sysupgrade.openwrt.org/api/v1/build/97314319da0e0e05498fedc5c37b607d679a220b127257e1ab22e5be6ea3962c` constantly return `{"status":202,"detail":"queued","queue_position":387,"enqueued_at":"2025-04-19T03:00:17.436804","request_hash":"97314319da0e0e05498fedc5c37b607d679a220b127257e1ab22e5be6ea3962c"}` , with `"queue_position"` staying stuck at `387`.

mwarning, 2025-04-19:
Hi, thank you for the details. I initially missed the URL in your screenshot. That should have been all I needed. Sorry for noise.

As for the error. Right now it is stuck at "Received build request". There might be @aparcar working on it.

mwarning, 2025-04-19:
Let's wait and I will try again later.

ronjouch, 2025-04-21:
Fixed! Thanks for your backstage work, thanks for OpenWrt! ❤

closed 2025-04-21 by ronjouch