openwrt/firmware-selector-openwrt-org #107: Build fails with server error

constgen, 2025-05-18:
https://firmware-selector.openwrt.org/?version=24.10.1&target=ramips%2Fmt7621&id=dlink_dir-2150-r1 . Request to build a custom preset fails with an error

[image]

[image]

lorand-horvath, 2025-05-19:
@aparcar Can you please take a look? The firmware-selector doesn't work still.

tmsBodnar, 2025-05-19:
[image]

swapnanil1, 2025-05-24:
It's still not working, and I can't figure out why. I tried running the server locally (using the master branch), and even tested with older versions (sources.zip), but the issue persists. My best guess is that something may have changed on the official OpenWrt server — perhaps a dependency, or maybe a refactor or restructure — and that change has broken the project.

mwarning, 2025-05-24:
Yes, something looks broken on the server side. There is nothing I ca do.

lorand-horvath, 2025-05-25:
https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721/1377

swapnanil1, 2025-05-26:
It's Fixed!!!! My thanks to the DEVs working for a better internet.

constgen, 2025-05-26:
Confirmed