openwrt/firmware-selector-openwrt-org #111: Can't build a custom image

LordArrin, 2025-06-02:
Traceback (most recent call last): File "/usr/local/lib/python3.12/site-packages/rq/worker.py", line 1633, in perform_job return_value = job.perform() ^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1331, in perform self._result = self._execute() ^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1365, in _execute result = self.func(*self.args, **self.kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/app/asu/build.py", line 293, in build container.kill() File "/usr/local/lib/python3.12/site-packages/podman/domain/containers.py", line 265, in kill response.raise_for_status() File "/usr/local/lib/python3.12/site-packages/podman/api/client.py", line 75, in raise_for_status raise APIError(cause, response=self._response, explanation=message) podman.errors.exceptions.APIError: 409 Client Error: Conflict (can only kill running containers. 49d4c78d5c72672ce25b2fa849bbd428a47f072f526a90683f4adf1b12157465 is in state stopped: container state improper)

My device: FriendlyARM NanoPi R3S

Packages: `base-files ca-bundle e2fsprogs firewall4 fstools kmod-gpio-button-hotplug kmod-nft-offload libc libgcc libustream-mbedtls logd mkf2fs mtd netifd nftables odhcp6c odhcpd-ipv6only opkg partx-utils ppp ppp-mod-pppoe procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd kmod-r8169 dnsmasq-full luci-ssl-nginx

liblua libubus libubus-lua libuci-lua lua luci-base luci-lib-ip luci-lib-jsonc luci-lib-nixio luci-mod-admin-full luci-theme-bootstrap rpcd luci-app-opkg luci-proto-ipv6 luci-proto-ppp luci-i18n-base-ru luci-i18n-firewall-ru luci-i18n-attendedsysupgrade-ru unzip curl bind-dig htop mc gawk grep sed coreutils-sort ca-certificates bind-host curl sudo lsof iptables-nft ip6tables-nft kmod-nfnetlink-queue kmod-nf-conncount kmod-nft-queue kmod-nft-xfrm kmod-nft-compat kmod-nft-connlimit luci-mod-rpc wget-ssl bridger ethtool

blkid blkdiscard block-mount busybox ca-certificates cgi-io coreutils coreutils-base64 coreutils-sleep coreutils-sort curl dropbear eip197-mini-firmware fwtool getrandom gre gzip ip-full iptables-mod-ipopt iw iwinfo jansson4 jq jshn jsonfilter  

kmod-asn1-decoder kmod-asn1-encoder kmod-atm kmod-br-netfilter kmod-cfg80211 kmod-crypto-acompress kmod-crypto-aead kmod-crypto-arc4 kmod-crypto-authenc kmod-crypto-ccm kmod-crypto-cmac kmod-crypto-crc32 kmod-crypto-crc32c kmod-crypto-ctr kmod-crypto-des kmod-crypto-ecb kmod-crypto-gcm kmod-crypto-geniv kmod-crypto-gf128 kmod-crypto-ghash kmod-crypto-hash kmod-crypto-hmac kmod-crypto-kpp kmod-crypto-lib-chacha20 kmod-crypto-lib-chacha20poly1305 kmod-crypto-lib-curve25519 kmod-crypto-lib-poly1305 kmod-crypto-manager kmod-crypto-md5 kmod-crypto-null kmod-crypto-rng kmod-crypto-seqiv kmod-crypto-sha1 kmod-crypto-sha256 kmod-crypto-sha3 kmod-crypto-sha512 kmod-crypto-user kmod-dnsresolver kmod-gpio-button-hotplug kmod-gre kmod-gre6 kmod-hwmon-core kmod-inet-diag kmod-ip6-tunnel kmod-ipt-core kmod-ipt-ipopt kmod-iptunnel kmod-iptunnel6 kmod-l2tp kmod-lib-crc-ccitt kmod-lib-crc16 kmod-lib-crc32c kmod-lib-textsearch kmod-libphy kmod-mdio-devres kmod-mii kmod-mppe kmod-nf-conntrack kmod-nf-conntrack-netlink kmod-nf-conntrack6 kmod-nf-flow kmod-nf-ipt kmod-nf-log kmod-nf-log6 kmod-nf-nat kmod-nf-nathelper kmod-nf-nathelper-extra kmod-nf-reject kmod-nf-reject6 kmod-nf-socket kmod-nf-tproxy kmod-nfnetlink kmod-nfnetlink-queue kmod-nft-core kmod-nft-fib kmod-nft-nat kmod-nft-offload kmod-nft-queue kmod-nft-socket kmod-nft-tproxy kmod-nls-base kmod-nls-cp1250 kmod-nls-cp437 kmod-nls-cp850 kmod-nls-iso8859-1 kmod-nls-iso8859-15 kmod-nls-ucs2-utils kmod-nls-utf8 kmod-oid-registry kmod-phy-realtek kmod-ppp kmod-pppoa kmod-pppoe kmod-pppol2tp kmod-pppox kmod-pptp kmod-scsi-core kmod-slhc kmod-thermal kmod-tls kmod-tun kmod-udptunnel4 kmod-udptunnel6 kmod-veth kmod-ipt-nat 

kmod-arptables kmod-crypto-cbc kmod-crypto-chacha20poly1305 kmod-crypto-cts kmod-crypto-deflate kmod-crypto-ecdh kmod-crypto-echainiv kmod-crypto-engine kmod-crypto-essiv kmod-crypto-fcrypt kmod-crypto-misc kmod-crypto-rsa kmod-cryptodev kmod-dax kmod-nfnetlink-cthelper kmod-nfnetlink-cttimeout kmod-nls-cp1251 

libatomic1 libblkid1 libblobmsg-json libbpf1 libc libcap libcares libcomerr0 libcurl4 libe2p2 libelf1 libev libevdev libext2fs2 libf2fs6 libgcc1 libgmp10 libiwinfo-data libiwinfo libjson-c5 libjson-script liblua liblucihttp-ucode liblucihttp0 libmnl0 libmount1 libncurses6 libnetfilter-conntrack3 libnettle8 libnfnetlink0 libnftnl11 libnghttp2-14 libnl-tiny1 libopenssl-conf libopenssl-legacy libopenssl3 libparted libplist libpthread libreadline8 librt libsmartcols1 libss2 libstdcpp6 libubox libubus-lua libubus libuci-lua libuci libuclient libucode libudebug libudev-zero libuuid1 libuv1 libwebsockets-full libxml2 libxtables12 linux-atm logd lsblk luci-i18n-opkg-ru luci-lib-ip luci-lib-jsonc luci-lib-nixio luci-mod-admin-full luci-mod-network luci-mod-status luci-mod-system luci-proto-ipv6 luci-proto-ppp luci-theme-bootstrap mkf2fs mtd netifd nftables-json odhcp6c odhcpd-ipv6only openssl-util openwrt-keyring opkg parted partx-utils ppp ppp-mod-pppoa ppp-mod-pppoe ppp-mod-pppol2tp ppp-mod-pptp procd procd-seccomp procd-ujail resolveip rpcd rpcd-mod-file rpcd-mod-iwinfo rpcd-mod-luci rpcd-mod-rrdns rpcd-mod-ucode terminfo uboot-envtools ubox ubus ubusd uci uclient-fetch ucode ucode-mod-fs ucode-mod-html ucode-mod-math ucode-mod-nl80211 ucode-mod-rtnl ucode-mod-ubus ucode-mod-uci ucode-mod-uloop urandom-seed urngd usign wget wget-ssl zlib

luci-i18n-ddns-ru e2fsprogs f2fs-tools fstools gawk grep htop kmod-fs-btrfs kmod-fs-exfat kmod-fs-ext4 kmod-fs-f2fs kmod-fs-ntfs3 kmod-fs-squashfs kmod-nf-conncount kmod-nfnetlink-queue kmod-nft-compat kmod-nft-connlimit kmod-nft-queue kmod-nft-xfrm kmod-usb-core kmod-usb-ehci kmod-usb-storage kmod-usb-storage-uas kmod-usb2 kmod-usb3 libc libgcc liblua libubus libubus-lua libuci-lua losetup lsof lua

kmod-usb-core kmod-usb-ehci kmod-usb-net kmod-usb-net-cdc-eem kmod-usb-net-cdc-ether kmod-usb-net-cdc-mbim kmod-usb-net-cdc-ncm kmod-usb-net-cdc-subset kmod-usb-net-huawei-cdc-ncm kmod-usb-net-ipheth kmod-usb-net-qmi-wwan kmod-usb-net-rndis kmod-usb-storage kmod-usb-wdm kmod-usb-xhci-hcd kmod-usb2 kmod-usb3 libusb-1.0-0 libusbmuxd usb-modeswitch usbids usbmuxd usbutils gdisk fstools

openssh-sftp-server openssh-server sshpass luci-i18n-filemanager-ru luci-i18n-dockerman-ru`

LordArrin, 2025-06-02:
yeah, my fault. luci-i18n-dockerman-ru too big

closed 2025-06-02 by LordArrin

mwarning, 2025-06-08:
We are working on better error messages.