openwrt/firmware-selector-openwrt-org #122: Docker Image

luckylinux, 2025-09-21:
It would be good for the Project to provide the Docker Image themselves, in case anybody (like me 😄) would like to self-host.

This is what I came up with (using my usual set of Scripts), with a Docker Image based on `nginx:latest`:
https://github.com/luckylinux/docker-firmware-selector-openwrt

I think however that it would probably be easier for somebody inside the Project to actively maintain it whenever a new Version is released.

Otherwise I hope this to be useful for other People that were in the same Situation as me 😉.

mwarning, 2025-09-21:
Hi there. Nice feat. But it is not hard (I hope) to set up the firmware selector. The dependencies are rather minimal (a web server).

mwarning, 2025-09-21:
Well, the point is that I do not have the motivation myself to maintain a docker setup for the firmware selector.

ToasterDEV, 2025-12-14:
> [@luckylinux](https://github.com/luckylinux) I'm curious—why are you want self-hosting it? It's just a static website, so the official version should work perfectly fine.

I'm not them, but a personal reason I'd like to be able to quickly spawn a container for would be to create images beyond the resource limits of the public workers.

I like to have access to the full toolset of some of the `shadow-*` packages instead of relying on BusyBox only, and after enough of them get added to a request it becomes increasingly likely the build will fail entirely.

I recognize that this is a niche use-case (as most routers barely have any storage in the first place), though for travel-intended units it can make a lot of stuff handier without carrying a lot of devices.

luckylinux, 2025-12-14:
> [@luckylinux](https://github.com/luckylinux) I'm curious—why are you want self-hosting it? It's just a static website, so the official version should work perfectly fine.

I wanted to be able to build everything locally.

Another Reason is to be "independent" in Case e.g. the Website goes down or anything.

In any Case, I want to have the flexibility to do so, which I believe should be part of any Open Source Project 😉.

CzBiX, 2025-12-14:
@luckylinux my project is a rewrite. so it requires its own Dockerfile.